princeton-vl/DROID-SLAM #1: cannot install on cudatoolkit<11.1 ？

dyy0205, 2021-09-02:
Thank you for your great work, but i cannot install 'droid_backends' on cudatoolkit=10.1，is it must be 11.1 according to environment.yaml？

zachteed, 2021-09-02:
I've tested with cuda 10.2 and 11.1, but I have not tried with 10.1. Could you post the error message you are getting when trying to compile?

dyy0205, 2021-09-03:
> I've tested with cuda 10.2 and 11.1, but I have not tried with 10.1. Could you post the error message you are getting when trying to compile?

thank you for your reply, it's a g++ compile error
[image]

jasonyzhang, 2022-02-23:
When I got the warning about Pytorch being compiled with g++ but c++ being the default compiler, I fixed it by doing this:
````
CC=gcc CXX=g++ python setup.py install
```

rmadan2, 2022-05-12:
j