princeton-vl/DROID-SLAM #2: Compiling Error

veinyyxy, 2021-09-03:
[image]
This command 'python setup.py install' report an error.

As shown above picture, the command of the red rectangle has reported an error. The error is followed picture.
[image]

By the way, My OS is window 10.

Thanks a lot.

hzwer, 2021-09-16:
I meet the same issue. Fortunately, I re-used cuda10.2 + cudnn8.0.4 + torch1.7.1 and successfully ran the demo.py provided by the author.

ZifaZhu, 2021-09-24:
> I meet the same issue using cuda11.0. Fortunately, I re-used cuda10.2 + cudnn8.0.4 + torch1.7.1 and successfully ran the demo.py provided by the author.

Which version of python did you use? Thx.

hzwer, 2021-09-24:
> > I meet the same issue using cuda11.0. Fortunately, I re-used cuda10.2 + cudnn8.0.4 + torch1.7.1 and successfully ran the demo.py provided by the author.
> 
> Which version of python did you use? Thx.

3.6

SongQiPing, 2023-04-02:
I think there is a problem with the Eigen version. VectorX is not available in Eigen 3.3, but it is available in VectorX 3.4. It is recommended to update the version of Eigen and build it from the source code on the official website