princeton-vl/DROID-SLAM #6: Loop Closure [Questions]

kbpachauri, 2021-09-08:
Hi,

Thanks for excellent work.   

1. In paper its mentioned "**If the camera returns to a previously mapped region, we add long range connections in the graph to perform loop closure."** Can you please tell me which portion of the code implement this? 

2. Is there a ablation studies which  you folks did on improvement (drift reduction) with loop closure? 

Best,
KP

Sanyam-Mehta, 2023-07-30:
Any updates on this? I haven't been able to find it either.

ThakurSarveshGit, 2024-08-07:
> Any updates on this? I haven't been able to find it either.

Loop closure happens in the back-end based on the reprojection error: https://github.com/princeton-vl/DROID-SLAM/issues/86#issuecomment-2270574977