princeton-vl/DROID-SLAM #7: Maybe typo in the cuda code of pose retraction

Willyzw, 2021-09-08:
Thanks for releasing this excellent work!
One small thing drew my attention and unsure if this is on purpose, which is the `xi[45]` at this line.
https://github.com/princeton-vl/DROID-SLAM/blob/9843482ec70a84d28defe3def4ed84b3a77ce914/src/droid_kernels.cu#L154

zachteed, 2021-09-08:
Thank you for pointing this out. This is indeed a bug and could cause an illegal memory access, but seems to have negligible impact on the accuracy of our system. And it only affects inference, not training or evaluation.

Anyways I am going to correct this issue. Our results may improve slightly after fixing the bug, and we are running some tests to check.

Willyzw, 2021-09-10:
Great!
Since you have correct the index in the last commit, I am closing this issue.

closed 2021-09-10 by Willyzw