princeton-vl/DROID-SLAM #9: bugs when compile

cuge1995, 2021-09-13:
```
: warning: expression has no effect

thirdparty/eigen/Eigen/src/Core/Product.h(145): catastrophic error: error while writing generated C file: No space left on device

1 catastrophic error detected in the compilation of "/home/ssd/zjl/zjl/slam/DROID-SLAM/thirdparty/lietorch/lietorch/src/lietorch_gpu.cu".
Compilation terminated.
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "python3.8/site-packages/torch/utils/cpp_extension.py", line 1666, in _run_ninja_build
    subprocess.run(
  File "python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['ninja', '-v']' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:
```

stanginch, 2021-11-02:
Hi,
I have encountered the same problem. Have you solved it?

cuge1995, 2021-11-11:
> Hi, I have encountered the same problem. Have you solved it?

not yet.