princeton-vl/DROID-SLAM #14: Question about the pose parsing in TartanAir dataset

xwjabc, 2021-10-11:
Hi, thank you for your great work! Recently when I read the data loading part of Droid-SLAM, I found the pose is parsed as:
```python
poses = np.loadtxt(osp.join(scene, 'pose_left.txt'), delimiter=' ')
poses = poses[:, [1, 2, 0, 4, 5, 3, 6]]   
```
From the [TartanAir tools](https://github.com/castacks/tartanair_tools/blob/master/data_type.md#pose-file), the line of the pose data file has the format `tx ty tz qx qy qz qw`, which uses a NED frame. In your implementation it seems you convert XYZ to YZX coordinates. I wonder if there is any reason behind it? Thanks!

closed 2021-10-13 by xwjabc

Rtut654, 2023-07-09:
@xwjabc  Hey! I know it is a bit offtop, but have you trained the model without depth, only with raw images and poses? This [issue](https://github.com/princeton-vl/DROID-SLAM/issues/10) states some troubles doing so. I wonder how to implement it if it's possible. Thanks in advance.

JingruiYu, 2025-02-17:
Thank you @zachteed for creating such a valuable resource!
Thank you @xwjabc for asking this `converting the poses` question and partially answering my confusion. 
But I still have a question,
To align axes XYZ→YZX, I assumed this requires left-multiplying each pose with the rotation matrix:  
   ```
   R_convert = np.array([
     0,0,1, # Original Z becomes new X
     1,0,0,  # Original X becomes new Y
     0,1,0   # Original Y becomes new Z
   ])
   ```
This should be a standard axis permutation.
   
But the above operation is directly adjusting the order of the vectors, which is fine on translation, but on rotation, you switch `qx,qy,qz` in the quaternions that represent the direction of rotation. The rotation is different before and after the switch. So this operation does not destroy the truth value? I'm not sure if I'm reading this wrong, or is this really a bug?