princeton-vl/DROID-SLAM #20: setup.py install failed

TBetterman, 2021-11-12:
Hi, when i run **python setup.py install**, the compile error happened: TypeError: expected string or bytes-like object

Python version: 3.8
setuptools version: 50.3.0

[image]

Please give some advice if convenient, thanks.

cuge1995, 2021-11-14:
same problem

snehahegdek, 2021-11-15:
I'm having the same problem.

TBetterman, 2021-11-15:
Hi, all, I have solved this problem @cuge1995 @snehahegdek :
- open the cooresponding **version.py**
- Changed the code on line 200 from **match = self._regex.search(version)** to **match = self._regex.search(str(version))**
- re-ran the code: **python setup.py install**

Enjoy!

Nekomiya-fubuki, 2021-11-16:
@TBetterman Hi,Can I ask a question?How can I open the cooresponding version.py.I can't find this program  erverywhere in the file.

doomzzju, 2021-11-22:
try python3 instead, man

zachteed, 2021-11-30:
This issue seemed to be caused by the PyTorch update. I've updated the environment.yaml file, which I believe solves the issue.

Sabrinashurong, 2022-01-19:
if one of the previous errors in the cuda compilation is "nvcc fatal   : Unsupported gpu architecture 'compute_80' \n ninja: build stopped: subcommand failed."the following solution may work. This is because that the mismatch between the expected computation power assignement in `setup.py` between the actual power in your computer.  So, if your computer gpu is RTX 2080 ti, the computation power parameters 8.0 and 8.6 can not be achieved. So, the solution of commenting the following lines (line 25, 26, 55, 56) in the `setup.py` may work.
```
#'-gencode=arch=compute_80,code=sm_80',
#'-gencode=arch=compute_86,code=sm_86',
```

Mikhail2017, 2024-02-20:
It could be also connected to the cpu version of torch package installed using environment.yaml. I was not able to install environment using conda, because it hanged on the resolving stage and used mamba instead, unfortunately it led to the incorrect package wo cuda support. Solved it finally installing all the packages separately using both conda and pip

closed 2024-03-05 by TBetterman