princeton-vl/DROID-SLAM #22: setup.py install fatal erro

Nekomiya-fubuki, 2021-11-17:
Hi,thanks for your code.I'm very interesting in it.But when I run the setup.py,I have a problem.
my driver:Nvidia 2080Ti
cuda:10.2
pytorch:1.10.0

running install
/home/u/anaconda3/envs/droidenv/lib/python3.7/site-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  setuptools.SetuptoolsDeprecationWarning,
/home/u/anaconda3/envs/droidenv/lib/python3.7/site-packages/setuptools/command/easy_install.py:159: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  EasyInstallDeprecationWarning,
running bdist_egg
running egg_info
writing droid_backends.egg-info/PKG-INFO
writing dependency_links to droid_backends.egg-info/dependency_links.txt
writing top-level names to droid_backends.egg-info/top_level.txt
/home/u/anaconda3/envs/droidenv/lib/python3.7/site-packages/torch/utils/cpp_extension.py:381: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
  warnings.warn(msg.format('we could not find ninja.'))
reading manifest file 'droid_backends.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'droid_backends.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_ext
building 'droid_backends' extension
gcc -pthread -B /home/u/anaconda3/envs/droidenv/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/u/Documents/RS/thirdparty/eigen -I/home/u/anaconda3/envs/droidenv/lib/python3.7/site-packages/torch/include -I/home/u/anaconda3/envs/droidenv/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/home/u/anaconda3/envs/droidenv/lib/python3.7/site-packages/torch/include/TH -I/home/u/anaconda3/envs/droidenv/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda-10.2/include -I/home/u/anaconda3/envs/droidenv/include/python3.7m -c src/droid.cpp -o build/temp.linux-x86_64-3.7/src/droid.o -O3 -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE="_gcc" -DPYBIND11_STDLIB="_libstdcpp" -DPYBIND11_BUILD_ABI="_cxxabi1011" -DTORCH_EXTENSION_NAME=droid_backends -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
/usr/local/cuda-10.2/bin/nvcc -I/home/u/Documents/RS/thirdparty/eigen -I/home/u/anaconda3/envs/droidenv/lib/python3.7/site-packages/torch/include -I/home/u/anaconda3/envs/droidenv/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/home/u/anaconda3/envs/droidenv/lib/python3.7/site-packages/torch/include/TH -I/home/u/anaconda3/envs/droidenv/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda-10.2/include -I/home/u/anaconda3/envs/droidenv/include/python3.7m -c src/droid_kernels.cu -o build/temp.linux-x86_64-3.7/src/droid_kernels.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options '-fPIC' -O3 -gencode=arch=compute_60,code=sm_60 -gencode=arch=compute_61,code=sm_61 -gencode=arch=compute_70,code=sm_70 -gencode=arch=compute_75,code=sm_75 -gencode=arch=compute_80,code=sm_80 -gencode=arch=compute_86,code=sm_86 -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE="_gcc" -DPYBIND11_STDLIB="_libstdcpp" -DPYBIND11_BUILD_ABI="_cxxabi1011" -DTORCH_EXTENSION_NAME=droid_backends -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
nvcc fatal   : Unsupported gpu architecture 'compute_80'
error: command '/usr/local/cuda-10.2/bin/nvcc' failed with exit status 1

How can I sovle it?Is it because the video card don't have enough computing power?

YunYi-Duo, 2021-11-17:
Did you solve it? I also have a problem. cuda10.0 pytorch1.10.0
[image]

Nekomiya-fubuki, 2021-11-17:
@YunYi-Duo Hi,I think this erro is that cuda10.0 does not support pytorch1.10.0.I remember that the highest version of pytorch support  cuda10.0 is 1.7.1. you can try to change the version of pytorch

YunYi-Duo, 2021-11-17:
> 
> 
> @YunYi-Duo Hi,I think this erro is that cuda10.0 does not support pytorch1.10.0.I remember that the highest version of pytorch support cuda10.0 is 1.7.1. you can try to change the version of pytorch

I think so. I am trying. thank you very much! ！

YunYi-Duo, 2021-11-17:
> 
> 
> @YunYi-Duo Hi,I think this erro is that cuda10.0 does not support pytorch1.10.0.I remember that the highest version of pytorch support cuda10.0 is 1.7.1. you can try to change the version of pytorch

nvcc fatal : Unsupported gpu architecture 'compute_80'
error: command '/usr/local/cuda100/bin/nvcc' failed with exit status 1

我现在也遇到这个问题了，你解决了吗？

Nekomiya-fubuki, 2021-11-17:
@YunYi-Duo   
I think it’s a problem with the graphics card’s computing power. For example, my 2080Ti video card can only support up to 75. The latter 80 and 86 are not supported, so I annotated it out, but I encountered a new problem and I didn’t know how to solve it
[image]
Actually，I'm not sure that I annotate this two lines of code is correct.
[image]

YunYi-Duo, 2021-11-18:
> @YunYi-Duo I think it’s a problem with the graphics card’s computing power. For example, my 2080Ti video card can only support up to 75. The latter 80 and 86 are not supported, so I annotated it [...]

我觉得是因为没有eigen。thirdparty文件夹下的eigen是空文件。应该是要自己装一下。我准备试试。

Nekomiya-fubuki, 2021-11-18:
Oh.thanks!You're right.It must download the eigen library, and just install directly with apt doesn't work

YunYi-Duo, 2021-11-18:
> 
> 
> Oh.thanks!You're right.It must download the eigen library, and just install directly with apt doesn't work

Did you run the demo successfully?This has a new problem.my cuda is 10.0.I am not sure if it is because of the cuda version.Can you give me a contact information to exchange?

Nekomiya-fubuki, 2021-11-18:
Sure,my qq number is 644745100

Nekomiya-fubuki, 2021-11-18:
> > Oh.thanks!You're right.It must download the eigen library, and just install directly with apt doesn't work
> 
> Did you run the demo successfully?This has a new problem.my cuda is 10.0.I am not sure if it is because of the cuda version.Can you give me a contact information to exchange?

Sure,How about QQ?I have a qq number is 644745100

closed 2021-11-18 by Nekomiya-fubuki