princeton-vl/DROID-SLAM #28: extact conda enviroment

CanCanZeng, 2021-12-10:
Hi, thank you for opening this great project!
When I try to install this project, I meet some compilation problem. It is probably related to conda environment, I find that there are also some other issues caused by env difference. So, I suggest you to create yaml by using `conda env export > environment.yml` command, so we can get the exact conda env.

CanCanZeng, 2021-12-13:
Finally I get it done. The environment.yaml produce by `conda env export > environment.yaml` on my computer is:
```
name: droid
channels:
  - https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
  - pytorch
  - https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
  - https://mirrors.tuna.tsinghua.edu.cn/anaconda/cloud/conda-forge
  - https://mirrors.ustc.edu.cn/anaconda/cloud/menpo/
  - https://mirrors.ustc.edu.cn/anaconda/cloud/bioconda/
  - https://mirrors.ustc.edu.cn/anaconda/cloud/msys2/
  - https://mirrors.ustc.edu.cn/anaconda/cloud/conda-forge/
  - https://mirrors.ustc.edu.cn/anaconda/pkgs/free/
  - https://mirrors.ustc.edu.cn/anaconda/pkgs/main/
  - https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free/
  - https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/pro/
  - https://mirrors.tuna.tsinghua.edu.cn/anaconda/cloud/pytorch/
  - https://mirrors.tuna.tsinghua.edu.cn/anaconda/cloud/conda-forge/
  - https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main/
  - defaults
dependencies:
  - _libgcc_mutex=0.1=conda_forge
  - _openmp_mutex=4.5=1_gnu
  - async_generator=1.10=py_0
  - attrs=21.2.0=pyhd8ed1ab_0
  - backcall=0.2.0=pyh9f0ad1d_0
  - backports=1.0=py_2
  - backports.functools_lru_cache=1.6.4=pyhd8ed1ab_0
  - blas=1.0=mkl
  - bleach=4.1.0=pyhd8ed1ab_0
  - bzip2=1.0.8=h7f98852_4
  - ca-certificates=2021.10.8=ha878542_0
  - cffi=1.15.0=py38h3931269_0
  - cudatoolkit=10.2.89=h8f6ccaa_9
  - debugpy=1.5.1=py38h709712a_0
  - decorator=5.1.0=pyhd8ed1ab_0
  - defusedxml=0.7.1=pyhd8ed1ab_0
  - entrypoints=0.3=pyhd8ed1ab_1003
  - ffmpeg=4.3=hf484d3e_0
  - freetype=2.10.4=h0708190_1
  - gmp=6.2.1=h58526e2_0
  - gnutls=3.6.13=h85f3911_1
  - importlib-metadata=4.8.2=py38h578d9bd_0
  - importlib_resources=5.4.0=pyhd8ed1ab_0
  - intel-openmp=2021.4.0=h06a4308_3561
  - ipykernel=6.6.0=py38he5a9106_0
  - ipython=7.30.1=py38h578d9bd_0
  - ipython_genutils=0.2.0=py_1
  - ipywidgets=7.6.5=pyhd8ed1ab_0
  - jbig=2.1=h7f98852_2003
  - jedi=0.18.1=py38h578d9bd_0
  - jinja2=3.0.3=pyhd8ed1ab_0
  - jpeg=9d=h36c2ea0_0
  - jsonschema=4.2.1=pyhd8ed1ab_1
  - jupyter_client=7.1.0=pyhd8ed1ab_0
  - jupyter_core=4.9.1=py38h578d9bd_1
  - jupyterlab_pygments=0.1.2=pyh9f0ad1d_0
  - jupyterlab_widgets=1.0.2=pyhd8ed1ab_0
  - lame=3.100=h7f98852_1001
  - lcms2=2.12=hddcbb42_0
  - ld_impl_linux-64=2.36.1=hea4e1c9_2
  - lerc=3.0=h9c3ff4c_0
  - libdeflate=1.8=h7f98852_0
  - libffi=3.4.2=h7f98852_5
  - libgcc-ng=11.2.0=h1d223b6_11
  - libgomp=11.2.0=h1d223b6_11
  - libiconv=1.16=h516909a_0
  - libnsl=2.0.0=h7f98852_0
  - libpng=1.6.37=h21135ba_2
  - libsodium=1.0.18=h36c2ea0_1
[... 98 lines cut ...]
    - nbclassic==0.3.4
    - open3d==0.14.1+5ac5fc4
    - opencv-python==4.5.4.60
    - pandas==1.3.4
    - pysocks==1.7.1
    - pytz==2021.3
    - pyyaml==6.0
    - requests==2.26.0
    - scikit-learn==1.0.1
    - scipy==1.7.3
    - seaborn==0.11.2
    - sniffio==1.2.0
    - soupsieve==2.3.1
    - threadpoolctl==3.0.0
    - tomlkit==0.7.2
    - torch-scatter==2.0.9
    - tqdm==4.62.3
    - urllib3==1.26.7
    - webencodings==0.5.1
    - websocket-client==1.2.3
prefix: /home/zcc/Software/Professional/anaconda3/envs/droid
```

My system is Ubuntu16.04, and since the open3d only support system latter then ubuntu18.04, so I build open3d from source. (you need to compile it with -DBUILD_GUI=OFF -DBUILD_WEBRTC=OFF)
gcc version is 7.4.0 (I meet compilation problem building extensions with default gcc5 )

darld, 2023-04-12:
when you facing problem when stuck on "solving environemtn" I suggest to replace your environment.yaml with the one above. For me it is worked.

Deng-King, 2023-04-22:
thx, my friend, that really helps.

Yaxun-Yang, 2023-05-18:
Really grateful! 
However I meet the following error
" Collecting package metadata (repodata.json): failed

CondaHTTPError: HTTP 000 CONNECTION FAILED for url <https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main/linux-64/repodata.json>
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.
'https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main/linux-64'

"
I solve this by changing channels with official version. Then my yaml file is 
`name: droid
channels:
  - rusty1s
  - pytorch
  - nvidia
  - conda-forge
  - defaults
dependencies:
  - _libgcc_mutex=0.1=conda_forge
  - _openmp_mutex=4.5=1_gnu
  - async_generator=1.10=py_0
  - attrs=21.2.0=pyhd8ed1ab_0
  - backcall=0.2.0=pyh9f0ad1d_0
  - backports=1.0=py_2
  - backports.functools_lru_cache=1.6.4=pyhd8ed1ab_0
  - blas=1.0=mkl
  - bleach=4.1.0=pyhd8ed1ab_0
  - bzip2=1.0.8=h7f98852_4
  - ca-certificates=2021.10.8=ha878542_0
  - cffi=1.15.0=py38h3931269_0
  - cudatoolkit=10.2.89=h8f6ccaa_9
  - debugpy=1.5.1=py38h709712a_0
  - decorator=5.1.0=pyhd8ed1ab_0
  - defusedxml=0.7.1=pyhd8ed1ab_0
  - entrypoints=0.3=pyhd8ed1ab_1003
  - ffmpeg=4.3=hf484d3e_0
  - freetype=2.10.4=h0708190_1
  - gmp=6.2.1=h58526e2_0
  - gnutls=3.6.13=h85f3911_1
  - importlib-metadata=4.8.2=py38h578d9bd_0
  - importlib_resources=5.4.0=pyhd8ed1ab_0
  - intel-openmp=2021.4.0=h06a4308_3561
  - ipykernel=6.6.0=py38he5a9106_0
  - ipython=7.30.1=py38h578d9bd_0
  - ipython_genutils=0.2.0=py_1
  - ipywidgets=7.6.5=pyhd8ed1ab_0
  - jbig=2.1=h7f98852_2003
  - jedi=0.18.1=py38h578d9bd_0
  - jinja2=3.0.3=pyhd8ed1ab_0
  - jpeg=9d=h36c2ea0_0
  - jsonschema=4.2.1=pyhd8ed1ab_1
  - jupyter_client=7.1.0=pyhd8ed1ab_0
  - jupyter_core=4.9.1=py38h578d9bd_1
  - jupyterlab_pygments=0.1.2=pyh9f0ad1d_0
  - jupyterlab_widgets=1.0.2=pyhd8ed1ab_0
  - lame=3.100=h7f98852_1001
  - lcms2=2.12=hddcbb42_0
  - ld_impl_linux-64=2.36.1=hea4e1c9_2
  - lerc=3.0=h9c3ff4c_0
  - libdeflate=1.8=h7f98852_0
  - libffi=3.4.2=h7f98852_5
  - libgcc-ng=11.2.0=h1d223b6_11
  - libgomp=11.2.0=h1d223b6_11
  - libiconv=1.16=h516909a_0
  - libnsl=2.0.0=h7f98852_0
  - libpng=1.6.37=h21135ba_2
  - libsodium=1.0.18=h36c2ea0_1
  - libstdcxx-ng=11.2.0=he4da1e4_11
  - libtiff=4.3.0=h6f004c6_2
  - libuv=1.42.0=h7f98852_0
  - libwebp-base=1.2.1=h7f98852_0
  - libzlib=1.2.11=h36c2ea0_1013
  - lz4-c=1.9.3=h9c3ff4c_1
  - markupsafe=2.0.1=py38h497a2fe_1
  - matplotlib-inline=0.1.3=pyhd8ed1ab_0
  - mistune=0.8.4=py38h497a2fe_1005
  - mkl=2021.4.0=h06a4308_640
  - mkl-service=2.4.0=py38h95df7f1_0
  - mkl_fft=1.3.1=py38h8666266_1
  - mkl_random=1.2.2=py38h1abd341_0
  - nbclient=0.5.9=pyhd8ed1ab_0
  - nbconvert=6.3.0=py38h578d9bd_1
  - nbformat=5.1.3=pyhd8ed1ab_0
  - ncurses=6.2=h58526e2_4
  - nest-asyncio=1.5.4=pyhd8ed1ab_0
  - nettle=3.6=he412f7d_0
  - notebook=6.4.6=pyha770c72_0
  - numpy=1.21.2=py38h20f2e39_0
  - numpy-base=1.21.2=py38h79a1101_0
  - olefile=0.46=pyh9f0ad1d_1
  - openh264=2.1.1=h780b84a_0
  - openjpeg=2.4.0=hb52868f_1
  - openssl=3.0.0=h7f98852_2
  - packaging=21.3=pyhd8ed1ab_0
  - pandoc=2.16.2=h7f98852_0
  - pandocfilters=1.5.0=pyhd8ed1ab_0
  - parso=0.8.3=pyhd8ed1ab_0
  - pexpect=4.8.0=pyh9f0ad1d_2
  - pickleshare=0.7.5=py_1003
  - pillow=8.4.0=py38h8e6f84c_0
  - pip=21.3.1=pyhd8ed1ab_0
  - prometheus_client=0.12.0=pyhd8ed1ab_0
  - prompt-toolkit=3.0.24=pyha770c72_0
  - ptyprocess=0.7.0=pyhd3deb0d_0
  - pycparser=2.21=pyhd8ed1ab_0
  - pygments=2.10.0=pyhd8ed1ab_0
  - pyparsing=3.0.6=pyhd8ed1ab_0
  - pyrsistent=0.18.0=py38h497a2fe_0
  - python=3.8.12=hf930737_2_cpython
  - python-dateutil=2.8.2=pyhd8ed1ab_0
  - python_abi=3.8=2_cp38
  - pytorch=1.10.0=py3.8_cuda10.2_cudnn7.6.5_0
  - pytorch-mutex=1.0=cuda
  - pyzmq=22.3.0=py38h2035c66_1
  - readline=8.1=h46c0cb4_0
  - send2trash=1.8.0=pyhd8ed1ab_0
  - setuptools=59.4.0=py38h578d9bd_0
  - six=1.16.0=pyh6c4a22f_0
  - sqlite=3.37.0=h9cd32fc_0
  - terminado=0.12.1=py38h578d9bd_1
  - testpath=0.5.0=pyhd8ed1ab_0
  - tk=8.6.11=h27826a3_1
  - torchaudio=0.10.0=py38_cu102
  - torchvision=0.11.1=py38_cu102
  - tornado=6.1=py38h497a2fe_2
  - traitlets=5.1.1=pyhd8ed1ab_0
  - typing_extensions=4.0.1=pyha770c72_0
  - wcwidth=0.2.5=pyh9f0ad1d_2
  - wheel=0.37.0=pyhd8ed1ab_1
  - widgetsnbextension=3.5.2=py38h578d9bd_1
  - xz=5.2.5=h516909a_1
  - zeromq=4.3.4=h9c3ff4c_1
  - zipp=3.6.0=pyhd8ed1ab_0
  - zlib=1.2.11=h36c2ea0_1013
  - zstd=1.5.0=ha95c52a_0
  - pip:
    - addict==2.4.0
    - anyio==3.4.0
    - argcomplete==1.12.3
    - argon2-cffi==21.3.0
    - argon2-cffi-bindings==21.2.0
    - babel==2.9.1
    - beautifulsoup4==4.10.0
    - certifi==2021.10.8
    - charset-normalizer==2.0.9
    - colorama==0.4.4
    - cycler==0.11.0
    - deprecation==2.1.0
    - droid-backends==0.0.0
    - evo==1.16.0
    - filelock==3.4.0
    - fonttools==4.28.3
    - gdown==4.2.0
    - idna==3.3
    - ipython-genutils==0.2.0
    - joblib==1.1.0
    - json5==0.9.6
    - jupyter-packaging==0.11.1
    - jupyter-server==1.13.1
    - jupyterlab==3.2.5
    - jupyterlab-server==2.8.2
    - kiwisolver==1.3.2
    - lietorch==0.2
    - matplotlib==3.5.1
    - natsort==8.0.1
    - nbclassic==0.3.4
    - open3d==0.14.1+5ac5fc4
    - opencv-python==4.5.4.60
    - pandas==1.3.4
    - pysocks==1.7.1
    - pytz==2021.3
    - pyyaml==6.0
    - requests==2.26.0
    - scikit-learn==1.0.1
    - scipy==1.7.3
    - seaborn==0.11.2
    - sniffio==1.2.0
    - soupsieve==2.3.1
    - threadpoolctl==3.0.0
    - tomlkit==0.7.2
    - torch-scatter==2.0.9
    - tqdm==4.62.3
    - urllib3==1.26.7
    - webencodings==0.5.1
    - websocket-client==1.2.3
prefix: /home/...../envs/droid`

Irikos, 2023-08-10:
Anyone that can provide the yaml export with visualization and with more recent versions of the libraries please? (preferably working on Ubuntu 22.04) 

I'm stuck on solving environment for quite a while.

ERGOWHO, 2023-10-03:
> Anyone that can provide the yaml export with visualization and with more recent versions of the libraries please? (preferably working on Ubuntu 22.04)
> 
> I'm stuck on solving environment for quite a while.

Hi I had the same issue. But I do not follow the above yaml. I just opened the original YAML and installed them one by one, I just have an issue when installing torch and torchaudio versions, and I go to official web to reinstall the correct version and it works for me now. 

conda install pytorch==1.12.1 torchvision==0.13.1 torchaudio==0.12.1 cudatoolkit=11.3 -c pytorch

byronsit, 2023-12-12:
here is ubuntu20.04 verison, I hope it helps

nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Feb__7_19:32:13_PST_2023
Cuda compilation tools, release 12.1, V12.1.66
Build cuda_12.1.r12.1/compiler.32415258_0

Tue Dec 12 23:49:20 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.125.06   Driver Version: 525.125.06   CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  N/A |
| 31%   38C    P2    44W / 320W |   9115MiB / 16376MiB |      4%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1365      G   /usr/lib/xorg/Xorg                235MiB |
|    0   N/A  N/A      1641      G   /usr/bin/gnome-shell               48MiB |
|    0   N/A  N/A      2909      G   ...481116074311298441,131072       99MiB |
|    0   N/A  N/A   1299151      G   ...AAAAAAAA== --shared-files       11MiB |
|    0   N/A  N/A   1482981      G   ...Tencent\WeChat\WeChat.exe       10MiB |
|    0   N/A  N/A   2073198      C   python                           8260MiB |
|    0   N/A  N/A   2073316    C+G   ...envs/DROID_new/bin/python      443MiB |
+-----------------------------------------------------------------------------+

name: DROID_new
channels:
  - defaults
dependencies:
  - _libgcc_mutex=0.1=main
  - _openmp_mutex=5.1=1_gnu
  - ca-certificates=2023.08.22=h06a4308_0
  - cudatoolkit=11.3.1=h2bc3f7f_2
  - ld_impl_linux-64=2.38=h1181459_1
  - libffi=3.4.4=h6a678d5_0
  - libgcc-ng=11.2.0=h1234567_1
  - libgomp=11.2.0=h1234567_1
  - libstdcxx-ng=11.2.0=h1234567_1
  - ncurses=6.4=h6a678d5_0
  - openssl=3.0.12=h7f8727e_0
  - pip=23.3.1=py38h06a4308_0
  - python=3.8.18=h955ad1f_0
  - readline=8.2=h5eee18b_0
  - setuptools=68.0.0=py38h06a4308_0
  - sqlite=3.41.2=h5eee18b_0
  - tk=8.6.12=h1ccaba5_0
  - wheel=0.41.2=py38h06a4308_0
  - xz=5.4.5=h5eee18b_0
  - zlib=1.2.13=h5eee18b_0
  - pip:
    - absl-py==2.0.0
    - addict==2.4.0
    - ansi2html==1.9.1
    - asttokens==2.4.1
    - attrs==23.1.0
    - backcall==0.2.0
    - beautifulsoup4==4.12.2
    - blinker==1.7.0
    - cachetools==5.3.2
    - certifi==2023.11.17
    - cffi==1.16.0
    - charset-normalizer==3.3.2
    - click==8.1.7
    - colorama==0.4.6
    - comm==0.2.0
    - configargparse==1.7
    - contourpy==1.1.1
    - cycler==0.12.1
    - dash==2.14.2
    - dash-core-components==2.0.0
    - dash-html-components==2.0.0
    - dash-table==5.0.0
    - decorator==5.1.1
    - droid-backends==0.0.0
    - evo==1.25.2
    - executing==2.0.1
    - fastjsonschema==2.19.0
    - flask==3.0.0
    - fonttools==4.46.0
    - gdown==4.7.1
    - google-auth==2.25.2
    - google-auth-oauthlib==1.0.0
    - grpcio==1.60.0
    - idna==3.6
    - importlib-metadata==7.0.0
    - importlib-resources==6.1.1
    - ipython==8.12.3
    - ipywidgets==8.1.1
    - itsdangerous==2.1.2
    - jedi==0.19.1
    - joblib==1.3.2
    - jsonschema==4.20.0
    - jsonschema-specifications==2023.11.2
    - jupyter-core==5.5.0
    - jupyterlab-widgets==3.0.9
    - kiwisolver==1.4.5
    - lietorch==0.2
    - lz4==4.3.2
    - markdown==3.5.1
    - matplotlib==3.7.4
    - matplotlib-inline==0.1.6
    - nbformat==5.7.0
    - nest-asyncio==1.5.8
    - oauthlib==3.2.2
    - open3d==0.17.0
    - opencv-python==4.8.1.78
    - packaging==23.2
    - parso==0.8.3
    - pexpect==4.9.0
    - pickleshare==0.7.5
    - pillow==10.1.0
    - pkgutil-resolve-name==1.3.10
    - platformdirs==4.1.0
    - plotly==5.18.0
    - pluggy==1.3.0
    - prompt-toolkit==3.0.42
    - protobuf==4.25.1
    - ptyprocess==0.7.0
    - pure-eval==0.2.2
    - pyasn1==0.5.1
    - pyasn1-modules==0.3.0
    - pycparser==2.21
    - pygments==2.17.2
    - pyparsing==3.1.1
    - pyquaternion==0.9.9
    - pysocks==1.7.1
    - pyyaml==6.0.1
    - referencing==0.32.0
    - requests==2.31.0
    - requests-oauthlib==1.3.1
    - retrying==1.3.4
    - rpds-py==0.13.2
    - rsa==4.9
    - scikit-learn==1.3.2
    - six==1.16.0
    - soupsieve==2.5
    - stack-data==0.6.3
    - suitesparse-graphblas==8.2.1.0
    - tenacity==8.2.3
    - tensorboard==2.14.0
    - tensorboard-data-server==0.7.2
    - threadpoolctl==3.2.0
    - torch-scatter==2.1.2
    - torchaudio==2.1.0
    - tqdm==4.66.1
    - traitlets==5.14.0
    - urllib3==2.1.0
    - wcwidth==0.2.12
    - werkzeug==3.0.1
    - widgetsnbextension==4.0.9
    - zipp==3.17.0
prefix: /home/xbh/miniconda3/envs/DROID_new

robofar, 2024-03-10:
I have issue when running demo.py. 
`python demo.py --imagedir=data/abandonedfactory --calib=calib/tartan.txt --stride=2`

Then image window is open, and it runs over images, and in the end it says:
`Traceback (most recent call last):
  File "/home/faris/Desktop/DROID-SLAM/demo.py", line 134, in <module>
    traj_est = droid.terminate(image_stream(args.imagedir, args.calib, args.stride))
  File "/home/faris/Desktop/DROID-SLAM/droid_slam/droid.py", line 81, in terminate
    self.backend(7)
  File "/home/faris/miniconda3/envs/droid/lib/python3.9/site-packages/torch/autograd/grad_mode.py", line 28, in decorate_context
    return func(*args, **kwargs)
  File "/home/faris/Desktop/DROID-SLAM/droid_slam/droid_backend.py", line 34, in __call__
    graph.add_proximity_factors(rad=self.backend_radius,
  File "/home/faris/Desktop/DROID-SLAM/droid_slam/factor_graph.py", line 374, in add_proximity_factors
    ii, jj = torch.as_tensor(es, device=self.device).unbind(dim=-1)
ValueError: not enough values to unpack (expected 2, got 0)
`

My specs are:
OS: ubuntu 20.04
GPU: 4GB VRAM

python=3.9.18
pytorch=1.10.1
cudatoolkit=11.3.1