princeton-vl/DROID-SLAM #29: How to create dense 3D point cloud

snehahegdek, 2021-12-12:
Hello, thank you for sharing your great work!

I'd like to create a dense 3D point cloud of the scene similar to the one in DeepV2D.

Could you help give me the steps and/or information I need to alter the code.

Thank you

HannahHaensen, 2022-03-21:
@snehahegdek hey any progress here, I would like to change the code to output update voxel maps

tomhruby1, 2022-05-10:
For exporting the point cloud I have hackishly edited the visualizer: https://github.com/tomhruby1/DROID-SLAM

pranav-asthana, 2022-08-17:
Since the visualization is working with highly downsampled images, how can we save full resolution point clouds?

XF522, 2024-05-04:
> For exporting the point cloud I have hackishly edited the visualizer: https://github.com/tomhruby1/DROID-SLAM

Thank you very much! Your visualizer works well in my system, which solve the problem of reconstruction.