princeton-vl/DROID-SLAM #32: setup.py problem

xhangHU, 2021-12-20:
src/droid_kernels.cu:15:10: fatal error: Eigen/Sparse: No such file or directory
 #include <Eigen/Sparse>
          ^~~~~~~~~~~~~~
compilation terminated.
error: command '/usr/local/cuda-11/bin/nvcc' failed with exit code 1

when i run python setup.py install, i got some problem, can you help me~

Riser6, 2021-12-20:
> src/droid_kernels.cu:15:10: fatal error: Eigen/Sparse: No such file or directory #include <Eigen/Sparse> ^~~~~~~~~~~~~~ compilation terminated. error: command '/usr/local/cuda-11/bin/nvcc' failed with exit code 1
> 
> when i run python setup.py install, i got some problem, can you help me~

It is possible that your third party (./thirdparty/eigen) dose not be downloaded

xhangHU, 2021-12-21:
I installed this, but this error is still reported..

jasonyzhang, 2022-02-23:
I had a similar problem that I solved because I did not clone with `--recursive`. Even though the Eigen folder was there, the submodule was not cloned.

lockingball, 2022-04-19:
Is Eigen needed download by ourself? i cannot find download part in code

rmadan2, 2022-05-25:
I solved this by using symbolic links to eigen3 in /usr/include

pradyumnakulkarni87, 2022-06-07:
> I had a similar problem that I solved because I did not clone with `--recursive`. Even though the Eigen folder was there, the submodule was not cloned.

pradyumnakulkarni87, 2022-06-07:
The detected CUDA version (10.1) mismatches the version that was used to compile
PyTorch (11.3). Please make sure to use the same CUDA versions.
 getting this error , while execution python setup.py install

pradyumnakulkarni87, 2022-06-07:
(DRD) itadmin:~/DROID-SLAM$ sudo python3 setup.py install
Traceback (most recent call last):
  File "setup.py", line 2, in <module>
    from torch.utils.cpp_extension import BuildExtension, CUDAExtension
ModuleNotFoundError: No module named 'torch'
But torch in installed , we can import torch 
Another Error:
(DRD) itadmin:~/DROID-SLAM$ python3 setup.py install
nvcc fatal   : Unsupported gpu architecture 'compute_80'
error: command '/bin/nvcc' failed with exit code 1

PLz help me

surajiitd, 2022-06-07:
@pradyumnakulkarni87  what is the output of `torch.cuda.get_arch_list()`? after importing torch.
you need to comment some lines in setup.py:
if your GPU doesn't support compute_86 and compute_80 (which seems to be your case), comment out line no: 25, 26, 55, 56.
I found that in one of the issues earlier.

pradyumnakulkarni87, 2022-06-08:
> For your 2nd error, after that it will work I think:
> 
> @pradyumnakulkarni87 what is the output of `torch.cuda.get_arch_list()`? after importing torch. you need to comment some lines in setup.py: if your GPU doesn't support compute_86 and compute_80 (which seems to be your case), comment out line no: 25, 26, 55, 56. I found that in one of the issues earlier.

Thank you ! It works after Commenting ....Thank you

chihyuanchiu, 2022-07-22:
Hi - When running setup.py, I received the following error. Would anyone happen to know what it is about? Thanks.

No CUDA runtime is found, using CUDA_HOME='/Users/chih-yuanchiu/opt/anaconda3'
running install
/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running bdist_egg
running egg_info
creating droid_backends.egg-info
writing droid_backends.egg-info/PKG-INFO
writing dependency_links to droid_backends.egg-info/dependency_links.txt
writing top-level names to droid_backends.egg-info/top_level.txt
writing manifest file 'droid_backends.egg-info/SOURCES.txt'
reading manifest file 'droid_backends.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'droid_backends.egg-info/SOURCES.txt'
installing library code to build/bdist.macosx-10.9-x86_64/egg
running install_lib
running build_ext
Traceback (most recent call last):
  File "setup.py", line 7, in <module>
    setup(
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/__init__.py", line 87, in setup
    return distutils.core.setup(**attrs)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 148, in setup
    return run_commands(dist)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 163, in run_commands
    dist.run_commands()
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 967, in run_commands
    self.run_command(cmd)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/dist.py", line 1214, in run_command
    super().run_command(command)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
    cmd_obj.run()
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/command/install.py", line 74, in run
    self.do_egg_install()
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/command/install.py", line 123, in do_egg_install
    self.run_command('bdist_egg')
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/dist.py", line 1214, in run_command
    super().run_command(command)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
    cmd_obj.run()
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/command/bdist_egg.py", line 165, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/command/bdist_egg.py", line 151, in call_command
    self.run_command(cmdname)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/dist.py", line 1214, in run_command
    super().run_command(command)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
    cmd_obj.run()
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/_distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/dist.py", line 1214, in run_command
    super().run_command(command)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
    cmd_obj.run()
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 79, in run
    _build_ext.run(self)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 410, in build_extensions
    self._check_cuda_version()
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 783, in _check_cuda_version
    torch_cuda_version = packaging.version.parse(torch.version.cuda)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/pkg_resources/_vendor/packaging/version.py", line 49, in parse
    return Version(version)
  File "/Users/chih-yuanchiu/opt/anaconda3/lib/python3.8/site-packages/pkg_resources/_vendor/packaging/version.py", line 264, in __init__
    match = self._regex.search(version)
TypeError: expected string or bytes-like object