princeton-vl/DROID-SLAM #33: download data error

Riser6, 2021-12-20:
Thanks for your great work! When I run the script "./tools/download_sample_data.sh".
In line 10 to line 12
```
gdown https://drive.google.com/uc?id=0B-ePgl6HF260NzQySklGdXZyQzA
unzip Barn.zip
rm Barn.zip
```
The url "https://drive.google.com/uc?id=0B-ePgl6HF260NzQySklGdXZyQzA" does not appear to exist. I would appreciate it that anyone can help me!

xhangHU, 2021-12-22:
I also encountered this problem, have you solved this problem?

Riser6, 2021-12-27:
No, the URL just doesn't exist

> I also encountered this problem, have you solved this problem?

psiorx, 2025-01-02:
This link seems to be working:
https://drive.google.com/file/d/0B-ePgl6HF260NzQySklGdXZyQzA/view?resourcekey=0-luQ7Jaym5BQL6IjxsgXY9A