princeton-vl/DROID-SLAM #34: CUDA error: no kernel image is available for execution on the device

junshutang, 2021-12-23:
Hi, thanks for the wonderful job!
However, I have met the error when I run the demo:
`python demo.py --imagedir=data/abandonedfactory --calib=calib/tartan.txt --stride=2`

The error happens after the first image is shown which is:
`1it [00:07,  7.25s/it]
Traceback (most recent call last):
  File "demo.py", line 102, in <module>
    droid.track(t, image, intrinsics=intrinsics)
  File "droid_slam/droid.py", line 66, in track
    self.filterx.track(tstamp, image, depth, intrinsics)
  File "/home/.conda/envs/py37/lib/python3.7/site-packages/torch/autocast_mode.py", line 198, in decorate_autocast
    return func(*args, **kwargs)
  File "/home/.conda/envs/py37/lib/python3.7/site-packages/torch/autograd/grad_mode.py", line 28, in decorate_context
    return func(*args, **kwargs)
  File "droid_slam/motion_filter.py", line 69, in track
    corr = CorrBlock(self.fmap[None,[0]], gmap[None,[0]])(coords0)
  File "droid_slam/modules/corr.py", line 47, in __call__
    corr = CorrSampler.apply(self.corr_pyramid[i], coords/2**i, self.radius)
RuntimeError: CUDA error: no kernel image is available for execution on the device
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/.conda/envs/py37/lib/python3.7/multiprocessing/spawn.py", line 105, in spawn_main
    exitcode = _main(fd)
  File "/home/.conda/envs/py37/lib/python3.7/multiprocessing/spawn.py", line 115, in _main
    self = reduction.pickle.load(from_parent)
  File "/home/.conda/envs/py37/lib/python3.7/multiprocessing/synchronize.py", line 110, in __setstate__
    self._semlock = _multiprocessing.SemLock._rebuild(*state)
FileNotFoundError: [Errno 2] No such file or directory
`

Could you help me figure it out?

BurhanUlTayyab, 2022-01-12:
It seems like a CUDA / Pytorch mismatch, perhaps try reinstalling CUDA or updating Pytorch to match CUDA's version.

pradyumnakulkarni87, 2022-06-07:
> It seems like a CUDA / Pytorch mismatch, perhaps try reinstalling CUDA or updating Pytorch to match CUDA's version.

What should be the least versions of CUDA and PyTorch .....plz help me out