princeton-vl/DROID-SLAM #36: Error by running Demos > FileNotFoundError:liblietorch.so

My-Corner, 2022-01-11:
Thanks for your great work. I have downloaded the model and I have faced an issue in running demos. when I run the first command "python demo.py --imagedir=data/abandonedfactory --calib=calib/tartan.txt --stride=2" I get an error "FileNotFoundError: Error: Could not find /home/cdh-h91/anaconda3/lib/python3.9/site-packages/lietorch/lib/liblietorch.so" I could not find liblietorch.so anywhere, I have installed lietorch... Any help would be appreciated!

YznMur, 2022-05-19:
any updates on this ?

YznMur, 2022-05-20:
use ``` pip install git+https://github.com/princeton-vl/lietorch.git ```

pradyumnakulkarni87, 2022-06-07:
ImportError: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /home/itadmin/anaconda3/envs/DRD_POC/lib/python3.9/site-packages/../../libopencv_gapi.so.405)

facing with this issue ...kindly help me out

pradyumnakulkarni87, 2022-06-09:
(DRD_POC) admin:~/DROID-SLAM$ python demo.py --imagedir=data/sfm_bench/rgb --calib=calib/eth.txt
0it [00:00, ?it/s]droid.pth
15it [00:06,  2.27it/s]Process Process-1:
Traceback (most recent call last):
  File "/home/itadmin/anaconda3/envs/DRD_POC/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/itadmin/anaconda3/envs/DRD_POC/lib/python3.9/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/itadmin/DROID-SLAM/droid_slam/visualization.py", line 145, in droid_visualization
    vis = o3d.visualization.VisualizerWithKeyCallback()
AttributeError: module 'open3d' has no attribute 'visualization'

Getting this error...plz help me out

pradyumnakulkarni87, 2022-06-09:
To save the reconstruction with full resolution depth maps use the --reconstruction_path flag.

where and how to use this

pradyumnakulkarni87, 2022-06-09:
Traceback (most recent call last):
  File "/home/itadmin/DROID-SLAM/demo.py", line 100, in <module>
    droid = Droid(args)
  File "/home/itadmin/DROID-SLAM/droid_slam/droid.py", line 37, in __init__
    from visualization import droid_visualization
  File "/home/itadmin/DROID-SLAM/droid_slam/visualization.py", line 8, in <module>
    import open3d as o3d
  File "/home/itadmin/anaconda3/envs/DRD1_POC/lib/python3.9/site-packages/open3d/__init__.py", line 9, in <module>
    from open3d.linux import *
  File "/home/itadmin/anaconda3/envs/DRD1_POC/lib/python3.9/site-packages/open3d/linux/__init__.py", line 7, in <module>
    globals().update(importlib.import_module('open3d.linux.open3d').__dict__)
  File "/home/itadmin/anaconda3/envs/DRD1_POC/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: /home/itadmin/anaconda3/envs/DRD1_POC/lib/python3.9/site-packages/open3d/linux/open3d.so: undefined symbol: _Py_ZeroStruct

Getting this error if we open3d is installed

pradyumnakulkarni87, 2022-06-10:
FileNotFoundError: [Errno 2] No such file or directory: 'data/Barn'
(DRD_POC) itadmin@GPU:~/DROID-SLAM$ python demo.py --imagedir=data/mav0/cam0/data --calib=calib/euroc.txt --t0=150
133it [00:00, 191.02it/s]droid.pth
167it [00:06, 12.07it/s]Process Process-1:
Traceback (most recent call last):
  File "/home/itadmin/anaconda3/envs/DRD_POC/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/itadmin/anaconda3/envs/DRD_POC/lib/python3.9/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/itadmin/DROID-SLAM/droid_slam/visualization.py", line 145, in droid_visualization
    vis = o3d.visualization.VisualizerWithKeyCallback()
AttributeError: module 'open3d' has no attribute 'visualization'
900it [04:23,  3.41it/s]
################################
Global BA Iteration #1
Global BA Iteration #2

above things are done using environment_novis.yaml

by executing one demo file getting above thing.....
following things making confusion 
1. what  Global BA iterations stand for.......what is us of it .
2.  what to do next......does the demo code helps for creation of VDO only on base of images .....
3. How can we get the 3D object from the given vdo...
4. What next step should be there 
KIndly guide and help

Yuchen-Song, 2022-10-06:
I guess you should consult at [lietorch oringin repo](https://github.com/princeton-vl/lietorch#installation) and follow their instructions on installation. However, I cannot successfully install it since it seems to need CUDA 10.2 to compile, but it is not supported by my system (Ubuntu 20.04)
[image]

lin-fangzhou, 2023-05-08:
> /home/cdh- h91/anaconda3/lib/python3.9/site-packages/lietorch/lib/liblietorch.so

really good suggestion

Rosa712, 2023-11-09:
> you can use `pip install git+https://github.com/princeton-vl/lietorch.git`
when I install it, error occurred.
[image]

rashikshrestha, 2024-09-03:
I was able to solve this by building `lietorch` from source following instructions [here](https://github.com/princeton-vl/lietorch?tab=readme-ov-file#installing-from-source)