princeton-vl/DROID-SLAM #39: Error on installing setp.py (Windows)

HannahHaensen, 2022-03-11:
**System:**
Windows 11
Python 3.9
Pytroch 1.10
cuda version 11.3
RTX 3070

lietorch installation works but droid_backend fails with:

```
droid_kernels.obj : error LNK2001: unresolved external symbol "public: long * __cdecl at::TensorBase::data_ptr<long>(void)const " (??$data_ptr@J@TensorBase@at@@QEBAPEAJXZ)
build\lib.win-amd64-3.9\droid_backends.cp39-win_amd64.pyd : fatal error LNK1120: 1 unresolved externals
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\link.exe' failed with exit code 1120
```

When I run setup.py to install droid_backend and lietroch I run into the a lot of warnings e.g.
```
C:/Users/hanna/Documents/GitHub/DROIDSLAM/venv/lib/site-packages/torch/include\c10/util/Optional.h(432): warning C4624: "c10::trivially_copyable_optimization_optional_base<T>": The destrcutor was deleted implicitly or defined as deleted
        with
        [
            T=std::vector<at::Tensor,std::allocator<at::Tensor>>
        ]
```

HannahHaensen, 2022-03-11:
already investigated this:
https://discuss.pytorch.org/t/mysterious-link-error-seen-by-multiple-people-while-building-cpp-cuda-extensions/91533/3

closed 2022-03-21 by HannahHaensen

hodakagoto, 2022-11-10:
Hi @HannahHaensen 
I am having the same problem.
> but ran into this, maybe it will work,

Can you tell me how to do that in detail?
Thank you.

denred0, 2023-02-02:
@hodakagoto just replace **long** with **int64_t** in the droid_kernels.cu file.

hodakagoto, 2023-02-02:
@denred0 
Thank you. Installation has been successful. I appreciate your help.