princeton-vl/DROID-SLAM #40: Training for 80000 times

jiesico, 2022-03-23:
Hi!
Thank you very much for sharing the code of DROID_SLAM.
I've trained 80,000 times on the Tartanair dataset so far, but why hasn't loss_function shown any signs of convergence so far?

fabiopoiesi, 2022-03-29:
Hi,

We are experiencing a loss like this during training Tartanair.

[image]

Is this something expected?

Thanks

xhangHU, 2022-05-03:
Can you use the weight file you got for testing? When I test with the weights I got from training, the following error is reported:
Traceback (most recent call last):
File "/root/docker2/droid/2new/DROID-SLAM/demo.py", line 117, in
traj_est = droid.terminate(image_stream(args.imagedir, args.calib, args.stride))
File "/root/docker2/droid/2new/DROID-SLAM/droid_slam/droid.py", line 81, in terminate
self.backend(7)
File "/root/anaconda3/envs/droidenv5/lib/python3.9/site-packages/torch/autograd/grad_mode.py", line 28, in decorate_context
return func(*args, **kwargs)
File "/root/docker2/droid/2new/DROID-SLAM/droid_slam/droid_backend.py", line 33, in call
graph.add_proximity_factors(rad=self.backend_radius,
File "/root/docker2/droid/2new/DROID-SLAM/droid_slam/factor_graph.py", line 368, in add_proximity_factors
ii, jj = torch.as_tensor(es, device=self.device).unbind(dim=-1)
ValueError: not enough values to unpack (expected 2, got 0)

realXiaohan, 2022-06-06:
@xhangHU Hi, I have run the model with the weights they provide successfully but met the same problem as I used my self-trained weights. Did u solve this problem yet?

xhangHU, 2022-06-06:
> @xhangHU Hi, I have run the model with the weights they provide successfully but met the same problem as I used my self-trained weights. Did u solve this problem yet?

It's not solved yet, and I checked that the network structure used in the pre-training model provided by the author is not the same as in the code provided

realXiaohan, 2022-06-06:
> It's not solved yet, and I checked that the network structure used in the pre-training model provided by the author is not the same as in the code provided
Hi, can I have your email please? We can talk about it in more details.

realXiaohan, 2022-06-06:
@xhangHU Hi, can I have your email please? We can talk about it in more detail.

lvmingzhe, 2022-06-07:
> @xhangHU Hi, can I have your email please? We can talk about it in more detail.

Hi, could I join yours? I would like to retrain the model but got stuck in the beginning. Thanks a lot.
12131040[at]mail.sustech.edu.cn

519174419, 2022-06-10:
> It's not solved yet, and I checked that the network structure used in the pre-training model provided by the author is not the same as in the code provided

It maybe cause by, the distance of graph is loss than the thresh. Try to give a smaller thresh, the demo will run successfully. But  I still meet some problem with the disps result

realXiaohan, 2022-06-10:
@519174419 Yeah, I solved the problem by setting a smaller thresh but the ego-motion prediction looks so weird for my self-trained model.

519174419, 2022-06-10:
> ### **[realXiaohan](/realXiaohan) ** commented [16分钟前](#issuecomment-1152353639)

Yes,I think the flow_loss and geo_loss is low, but it still something wrong with the disparity map.

realXiaohan, 2022-06-10:
@519174419 We can talk about it in more detail and my email is lixh@mail.ustc.edu.cn.

realXiaohan, 2022-06-10:
@519174419  We can talk about it in more detail and my email is [li2xh@mail.ustc.edu.cn].

YznMur, 2022-06-15:
> @519174419 Yeah, I solved the problem by setting a smaller thresh but the ego-motion prediction looks so weird with my self-trained model. Is anything wrong with your disparity map?

Hi @realXiaohan , could you please share your config for training and demo ?

Aurora-chevalier, 2023-03-08:
Hello, I can't read the image when training the tartan dataset, can you show the instructions you used to train the tartan dataset
[image]

cjd24-coder, 2023-07-21:
> > ### **[realXiaohan](/realXiaohan) ** commented [16分钟前](#issuecomment-1152353639)
> 
> Yes,I think the flow_loss and geo_loss is low, but it still something wrong with the disparity map.

Hello, I have been stuck here for a long time now, may I ask if you have solved this problem? If it is convenient, can you add a contact information for detailed communication? I would like to ask you about my wechat 18186416709, email 1063062177@qq.com

cjd24-coder, 2023-07-21:
> @519174419 Yeah, I solved the problem by setting a smaller thresh but the ego-motion prediction looks so weird with my self-trained model. Is anything wrong with your disparity map?
Hello, I have been stuck here for a long time now, may I ask if you have solved this problem? If it is convenient, can you add a contact information for detailed communication? I would like to ask you about my wechat 18186416709, email 1063062177@qq.com