princeton-vl/DROID-SLAM #43: After training, the ckpt flie can not use?

xhangHU, 2022-04-24:
0it [00:00, ?it/s]./checkpoints/bla_250000.pth
/root/anaconda3/envs/droidenv5/lib/python3.9/site-packages/torch/functional.py:445: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  /opt/conda/conda-bld/pytorch_1639180487213/work/aten/src/ATen/native/TensorShape.cpp:2157.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
51it [00:06,  7.47it/s]
################################
Traceback (most recent call last):
  File "/root/docker2/droid/2new/DROID-SLAM/demo.py", line 117, in <module>
    traj_est = droid.terminate(image_stream(args.imagedir, args.calib, args.stride))
  File "/root/docker2/droid/2new/DROID-SLAM/droid_slam/droid.py", line 81, in terminate
    self.backend(7)
  File "/root/anaconda3/envs/droidenv5/lib/python3.9/site-packages/torch/autograd/grad_mode.py", line 28, in decorate_context
    return func(*args, **kwargs)
  File "/root/docker2/droid/2new/DROID-SLAM/droid_slam/droid_backend.py", line 33, in __call__
    graph.add_proximity_factors(rad=self.backend_radius,
  File "/root/docker2/droid/2new/DROID-SLAM/droid_slam/factor_graph.py", line 368, in add_proximity_factors
    ii, jj = torch.as_tensor(es, device=self.device).unbind(dim=-1)
ValueError: not enough values to unpack (expected 2, got 0)
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/root/anaconda3/envs/droidenv5/lib/python3.9/multiprocessing/spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "/root/anaconda3/envs/droidenv5/lib/python3.9/multiprocessing/spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
  File "/root/anaconda3/envs/droidenv5/lib/python3.9/multiprocessing/synchronize.py", line 110, in __setstate__
    self._semlock = _multiprocessing.SemLock._rebuild(*state)
FileNotFoundError: [Errno 2] No such file or directory
[W CudaIPCTypes.cpp:15] Producer process has been terminated before all shared CUDA tensors released. See Note [Sharing CUDA tensors]

YznMur, 2022-06-08:
Hi @xhangHU
Have you managed to solve this problem?

519174419, 2022-06-10:
It maybe cause by, the distance of graph is loss than the thresh. Try to give a smaller thresh, the demo will run successfully. But I still meet some problem with the disps result

YznMur, 2022-06-10:
Hi @519174419
you mean keyframe_thresh?
can you please share your demo (inference) parameters?

xhangHU, 2022-06-10:
> It maybe cause by, the distance of graph is loss than the thresh. Try to give a smaller thresh, the demo will run successfully. But I still meet some problem with the disps result

do you mean n_frames in the parameter, i tried 3 and 4 and it doesn't work

519174419, 2022-06-13:
> 

> Hi @519174419 would you please share your demo (inference) parameters or training parameters so the ckpt file could work ?
It's still some problem with my result, I am trying to solve it.

Aurora-chevalier, 2023-03-08:
Hello, I can't read the image when training the tartan dataset, can you show the instructions you used to train the tartan dataset？
[image]