princeton-vl/DROID-SLAM #46: Implementing on Jetson Xavier NX

rmadan2, 2022-05-30:
I would like to begin by saying that this is indeed an exceptional effort. I am impressed by the robustness and stability of DROID-SLAM in particular. I am currently trying to implement DROID-SLAM on a Jetson Xavier NX. I understand that a GPU with at least 11GB of memory is needed for inference, but I am not currently interested in real-time performance. The Jetson Xavier NX board that I am using currently is running with Python 3.6, with CUDA 10.2 and Pytorch 1.10, and I am getting the following error:
[image]

rmadan2, 2022-05-30:
I understand that an 801 error means that the service isn't currently supported on the NVIDIA device. but is there a work-around to it?

surajiitd, 2022-06-05:
@rmadan2  did you resolved this issue?, I also got the same problem in Jetson Xavier AGX. Could you please help!

surajiitd, 2022-06-05:
And when I passed `--disable_vis` flag, then the new error is like this: 
`RuntimeError: CUDA error: no kernel image is available for execution on the device` 

My output for `torch.cuda.is_available()` is  `True`

My output for `torch.cuda.get_arch_list()` is  `['sm_53', 'sm_62', 'sm_72']`

My output for `torch.Tensor([10,23]).to("cuda")` is `tensor([10., 23.], device='cuda:0')`

[image]

rmadan2, 2022-06-07:
Hi @surajiitd , I have not been able to resolve this issue

chaitanya100100, 2023-05-06:
@surajiitd did you manage to solve this kernel image error?

surajiitd, 2023-05-06:
Yes, you need to give appropriate compiler flags for [line no. 21](https://github.com/princeton-vl/DROID-SLAM/blob/8016d2b9b72b101a3e9ac804ebf20b4c654dc291/setup.py#L21) and 51, instead of compiling for those 6 architectures, you need to do that for ARM architecture.
I don't remember properly, for jetson orin it was '-march=armv8-a' I guess.

chaitanya100100, 2023-05-09:
Thanks !! You saved many hours of misery for me. My GPU was very old for the default architectures mentioned on those lines. I had to add _50 flags.

TryTwoTop, 2024-01-01:
@surajiitd Hello.
Did you add only '-march=armv8-a' as the cxx option and '-gencode=arch=compute_72,code=sm_72' as the nvcc option to the setup.py file?

I wonder if I only need to modify setup.py .

surajiitd, 2024-01-29:
@TryTwoTop Sorry, I don't remember. You can share if you managed to resolve the issue.

surajiitd, 2024-01-31:
`'-gencode=arch=compute_87,code=sm_87'` line worked in Jetson AGX Orin (it has Ampere architecture) to compile `lietorch` and `droid_backends`.

TryTwoTop, 2024-01-31:
@surajiitd Thank you so much for your precious time.