princeton-vl/DROID-SLAM #48: DEPTH_SCALE value

YznMur, 2022-06-03:
Thanks for your open source code
I am trying to train the model on KITTI. So I want to ask, how DEPTH_SCALE value is decided? currently, for tartan, it's 
set to 5

CH-901, 2023-02-06:
It should be 256

cjd24-coder, 2023-07-21:
> Thanks for your open source code I am trying to train the model on KITTI. So I want to ask, how DEPTH_SCALE value is decided? currently, for tartan, it's set to 5

您好，我现在也在这里卡了很久，请问您解决这个问题了吗，方便的话可以加个联系方式详细交流一下吗，想请教您一下，我的微信18186416709，邮箱1063062177@qq.com

cjd24-coder, 2023-07-21:
> Thanks for your open source code I am trying to train the model on KITTI. So I want to ask, how DEPTH_SCALE value is decided? currently, for tartan, it's set to 5

Hello, I have been stuck here for a long time now, may I ask if you have solved this problem? If it is convenient, can you add a contact information for detailed communication? I would like to ask you about my wechat 18186416709, email 1063062177@qq.com

NOTGOOOOD, 2023-10-24:
> Thanks for your open source code I am trying to train the model on KITTI. So I want to ask, how DEPTH_SCALE value is decided? currently, for tartan, it's set to 5

I'm also confused about this.