princeton-vl/DROID-SLAM #54: Essential Libraries Version Requirements

pradyumnakulkarni87, 2022-06-14:
Requested to kindly provide the least or exact version required for following
Python
Open3d
Pytorch

pradyumnakulkarni87, 2022-06-14:
Getting error if Pytorch with 1.10 
(DRD_POC) itadmin@NSK-GPU:~/DROID-SLAM$ python demo.py --imagedir=data/sfm_bench/rgb --calib=calib/eth.txt
Traceback (most recent call last):
  File "/home/itadmin/DROID-SLAM/demo.py", line 7, in <module>
    import lietorch
  File "/home/itadmin/anaconda3/envs/DRD_POC/lib/python3.9/site-packages/droid_backends-0.0.0-py3.9-linux-x86_64.egg/lietorch/__init__.py", line 2, in <module>
    from .groups import LieGroupParameter, SO3, RxSO3, SE3, Sim3, cat, stack
  File "/home/itadmin/anaconda3/envs/DRD_POC/lib/python3.9/site-packages/droid_backends-0.0.0-py3.9-linux-x86_64.egg/lietorch/groups.py", line 5, in <module>
    from .group_ops import Exp, Log, Inv, Mul, Adj, AdjT, Jinv, Act3, Act4, ToMatrix, ToVec, FromVec
  File "/home/itadmin/anaconda3/envs/DRD_POC/lib/python3.9/site-packages/droid_backends-0.0.0-py3.9-linux-x86_64.egg/lietorch/group_ops.py", line 1, in <module>
    import lietorch_backends
ImportError: /home/itadmin/anaconda3/envs/DRD_POC/lib/python3.9/site-packages/droid_backends-0.0.0-py3.9-linux-x86_64.egg/lietorch_backends.cpython-39-x86_64-linux-gnu.so: undefined symbol: _ZNK2at6Tensor7optionsEv

Kindly help me out

alik-git, 2024-02-29:
+1 I would like this too