princeton-vl/DROID-SLAM #56: Execution after Reconstruction path files

ChinmayBH, 2022-06-22:
Hello! Thank you so much for this open source repository. Whole repo is working but I am not understanding how to use files that been generated and saved at reconstruction path. Those are 5 numpy files. Can any one guide how to use those files to create final output? that would be really useful ! Thanks again!

MrCorsair3, 2022-10-05:
Hello, do you found solution ?

senhuangpku, 2022-10-16:
Hello! I have the same problem, do you find the solution?

ChinmayBH, 2023-06-09:
yes. We don't want to use those files manually anywhere. Just a suggestion that create a requirment.txt first and then install that before proceeding to the actual execution of code. that will help you to smooth execution.

ChinmayBH, 2023-06-09:
I am closing this issue because the above mentioned method helped me to resolve issue.

closed 2023-06-09 by ChinmayBH

SidneyRey, 2023-08-02:
I have the same question, anyone can visualize these npy files?

ERGOWHO, 2023-10-31:
> I have the same question, anyone can visualize these npy files?

Hi have you solved the .npy problem yet? Thanks in advance