princeton-vl/DROID-SLAM #62: Inference error on windows

lqxisok, 2022-07-27:
I have built the extensions and then attempt to run a demo of `sfm_bench` provided at the readme. But after several tries, I always get an error with the following message:
```
ii, jj = torch.as_tensor(es, device=self.device).unbind(dim=-1)
ValueError: not enough values to unpack (expected 2, got 0)
````
An interesting observation is the images of the `sfm_bench` demo could be shown usually. While, when entering into the `driod.terminate()` function, the program crashes with the above error. I find that the value of `delta.norm(dim=-1).mean().item()` (in `motion_filter.py`) is strange ( ~0.0027 ). As the filter_threshold is set at about 2.4, I guess the frame data can not be added into `self.video`

My question is does it only happen on windows? I will give try on ubuntu to take a further test. Thx for reply.

songlin, 2023-01-06:
@lqxisok Hi, I met the same problem, Can you share how you solved this problem?

Sebastian-Garcia, 2023-12-12:
Was this ever solved on Windows?

wza527, 2025-04-19:
Hello, I am also experiencing the same problem, have you found a workaround?