princeton-vl/DROID-SLAM #63: Problem with reconstruction result

lilly5791, 2022-07-29:
Thank you for sharing such a great deep learning slam algorithm.
I also thank you for the recently modified code which outputs the numpy pose results on reconstruction_path.

It works very well on my own data and the demo result is almost same with the groundtruth!
However, I found a problem when I visualize the npy result with Matlab

As you can see in picture,
sfm_bench and rgbd_dataset_freiburg3_cabinet trajectory visualization result is same with demo visualization.
However my own data trajectory visualization result and the demo visualization is very different.

In my own data, the demo result is very good, but I don't understand why the matlab visualization result is bad even though the matlab code is same. 
Do you know why the npy result is different from the demo?

[image]

buenos-dan, 2022-08-02:
Would you like to leave a contact method with me?  I have some details to talk about. WeChat or E-mail?

buenos-dan, 2022-08-02:
I test EuRoc dataset, using reconstruction result to plot trajectory, it has a bad performance. Maybe sfm_bench and rgbd_dataset_freiburg3_cabinet dataset have a large co-vision relationship, and EuRoc and your own dataset do not. So reconstruction is bad. I wonder If you can record a custom dataset like walk around a desk and test again?

buenos-dan, 2022-08-02:
Yes, I have the same problem with you.
[image]

lilly5791, 2022-08-04:
Here's my email
1913084@sookmyung.ac.kr

The movement is different from yours and mine but the droid slam result looks similar especially the round and noisy part.

buenos-dan, 2022-08-09:
Try to save data directly from visualization.py and the result is good. Why reconstruction is bad? I guess poses were optimized after, so you should save the data in time and update data with dirty_index. PS: don't save tensor directly, use tensor.item() to save the value.

lilly5791, 2022-08-12:
Thank  you for your advice.
This worked well!

pranav-asthana, 2022-08-17:
Hi can you please explain what you mean by save the data in time and update data with dirty_index?

lilly5791, 2022-08-24:
I think it's because the type of video.poses is torch.tensor and the pose information keeps changing. I'm not used to torch so I just saved all ix and pose as txt file and only used last lines for visualization.

senhuangpku, 2022-10-16:
Hello, I have no idea about how to use the result of reconstruction_path, could you tell me your solution? Thanks!

pranav-asthana, 2022-10-16:
This is what I am using. Just leaving this here in case it helps anyone.
1. images, poses and depths are output for keyframes into reconstruction_path. These can be used with a reconstruction algorithm (like tsdf, a good implementation is in Open3D) or any other MVS system.
2. In demo.py, traj_est stores the pose for each frame of the input video after global refinement and trajectory filling. If you need this information, you can save it to allow doing things like MVS on each input frame.

RitvikMandyam, 2023-04-04:
> This is what I am using. Just leaving this here in case it helps anyone.
>
> 1. images, poses and depths are output for keyframes into reconstruction_path. These can be used with a reconstruction [...]

Could you share the code to do this, please? I'm working on a college project that uses DROID-SLAM, and I'd really like to show surfaces for my demo rather than a point cloud. I'm able to get pretty decent output by increasing filter_threshold in droid_slam/visualization.py, but it still looks terrible when you zoom into any part of the visualization.

surajiitd, 2023-06-11:
@pranav-asthana Do you know why the reconstructions are saved before global-BA ?
https://github.com/princeton-vl/DROID-SLAM/blob/8016d2b9b72b101a3e9ac804ebf20b4c654dc291/demo.py#L131-L134