princeton-vl/DROID-SLAM #64: setup.py problem

wangsc18, 2022-08-04:
subprocess.CalledProcessError: Command '['ninja', '-v']' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/data/wsc/data/DROID-SLAM/setup.py", line 7, in <module>
    setup(
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/__init__.py", line 87, in setup
    return distutils.core.setup(**attrs)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 185, in setup
    return run_commands(dist)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
    dist.run_commands()
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 973, in run_commands
    self.run_command(cmd)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/dist.py", line 1217, in run_command
    super().run_command(command)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 992, in run_command
    cmd_obj.run()
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/command/install.py", line 74, in run
    self.do_egg_install()
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/command/install.py", line 123, in do_egg_install
    self.run_command('bdist_egg')
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 319, in run_command
    self.distribution.run_command(command)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/dist.py", line 1217, in run_command
    super().run_command(command)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 992, in run_command
    cmd_obj.run()
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/command/bdist_egg.py", line 165, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/command/bdist_egg.py", line 151, in call_command
    self.run_command(cmdname)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 319, in run_command
    self.distribution.run_command(command)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/dist.py", line 1217, in run_command
    super().run_command(command)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 992, in run_command
    cmd_obj.run()
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/_distutils/command/install_lib.py", line 112, in build
    self.run_command('build_ext')
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 319, in run_command
    self.distribution.run_command(command)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/dist.py", line 1217, in run_command
    super().run_command(command)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 992, in run_command
    cmd_obj.run()
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/command/build_ext.py", line 79, in run
    _build_ext.run(self)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 346, in run
    self.build_extensions()
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/torch/utils/cpp_extension.py", line 735, in build_extensions
    build_ext.build_extensions(self)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 466, in build_extensions
    self._build_extensions_serial()
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 492, in _build_extensions_serial
    self.build_extension(ext)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/command/build_ext.py", line 202, in build_extension
    _build_ext.build_extension(self, ext)
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/setuptools/_distutils/command/build_ext.py", line 547, in build_extension
    objects = self.compiler.compile(
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/torch/utils/cpp_extension.py", line 556, in unix_wrap_ninja_compile
    _write_ninja_file_and_compile_objects(
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/torch/utils/cpp_extension.py", line 1399, in _write_ninja_file_and_compile_objects
    _run_ninja_build(
  File "/data/wsc/anaconda3/envs/droidenv_low/lib/python3.9/site-packages/torch/utils/cpp_extension.py", line 1733, in _run_ninja_build
    raise RuntimeError(message) from e
RuntimeError: Error compiling objects for extension

when i run setup.py, i met this problem, can you help me plz?

sdurand777, 2022-10-14:
I got the same error. Did you find a solution ?
Thank you

yanfeng-tang, 2023-07-18:
> 

I got the same problem. Can you help me?
Thank you

Sebastian-Garcia, 2023-12-12:
I don't recall exactly what fixed it for me, but I installed ninja, and had to do some environment variable manipulation with the 'cl.exe' file because this build error was related to that for me.

Daydreamer-f, 2024-07-31:
@wangsc18 @sdurand777 @yanfeng-tang I met the same problem, did you find a solution? Thank you.

Daydreamer-f, 2024-07-31:
> @wangsc18 @sdurand777 @yanfeng-tang I met the same problem, did you find a solution? Thank you.

Hi guys, I know how to solve this problem. It is still a problem with version of cuda and pytorch. At first, I use cuda-12.1 and torch 2.2.0, and met this problem. So I create a new environment with cuda-11.3 and torch 1.10 (exactly the same as the readme required), and it worked.

Ariesone614, 2024-12-09:
> Hi guys, I know how to solve this problem. It is still a problem with version of cuda and pytorch. At first, I use cuda-12.1 and torch 2.2.0, and met this problem. So I create a new environment with [...]

哥们，你是把之前本地下载的cuda删了，又重新下的吗？

Daydreamer-f, 2024-12-10:
> 哥们，你是把之前本地下载的cuda删了，又重新下的吗？

集群上有多个cuda version，我不是在本地跑的