princeton-vl/DROID-SLAM #66: Stereo+RGBD input

felipesce, 2022-08-18:
Hello everyone.

I have a stereo setup and feed an RGBD stream to DROID with great results.
Thank you very much for this piece of software.

My question is if it's possible to feed both images from the stereo rig (left and right).

I tried defining args.stereo=True but couldn't quite understand how to name the images so that they are fed in the right order.

Thank you all for your time and work.

Cheers.

closed 2022-08-24 by felipesce

gyes00205, 2022-12-27:
Hi @felipesce,
Have you solved this problem how to feed the stereo sequence into droid-slam?
Thanks.

spsingh37, 2024-12-31:
Hi @gyes00205 @felipesce , how were you able to run DROID-SLAM with stereo sequence? I tried running it but the trajectory scales up really bad over time. Would really appreciate any help.

Thanks