princeton-vl/DROID-SLAM #68: Training with very different data

felipesce, 2022-09-08:
Hello there,

Firstly thanks again to everyone making this software possible. Amazing stuff.

I'm trying to train on a personal dataset of very tiny things looked up close.

When I train with the dafault parameters, then the model overfits to only correct translations, not rotations. This happens both when training from scratch and when starting from the droid.pth checkpoint.

Rotations will still be of the same magnitude, because they are constrained to the (-1,1) range, but translations have me a bit confused.
I'm guessing the pose and depth values in tartan are in meters? would there be trouble training with a dataset in millimeters?

I tried playing around with the DEPTH_SCALE value but no improvement still.

I have many questions and ideas, there are many knobs to move, but maybe some parameter is amiss.
What effect would it have changing the fmin and fmax for example?

Thanks for any and all help and insights you can move my way.

felipesce, 2022-09-12:
Found a DEPTH_SCALE value that worked, in my case it was 0.1

closed 2022-09-12 by felipesce

zhangjd1029, 2023-06-26:
Hello, I also want to train my own dataset, but I'm not sure what data is needed. Can you share which dataset you used during training and how to place the data? Thank you.

cjd24-coder, 2023-12-21:
> Found a DEPTH_SCALE value that worked, in my case it was 0.1

hello! Thank you very much for your question. I also encountered the same problem during training. Can I discuss it with you in detail? My email is 1063062177@qq.com and WeChat is 18186416709. Thank you very much.

cjd24-coder, 2023-12-21:
> Hello there,
>
> Firstly thanks again to everyone making this software possible. Amazing stuff.
>
> I'm trying to train on a personal dataset of very tiny things looked up close.
>
> When I train with the [...]
hello! Thank you very much for your question. I also encountered the same problem during training. Can I discuss it with you in detail? My email is 1063062177@qq.com and WeChat is 18186416709. Thank you very much.