princeton-vl/DROID-SLAM #78: Method for giving rgbd data as input

SuruchiPokhrel, 2022-11-08:
I have been using **demo.py** with custom dataset and the results are sparse. DROID-SLAM is supposed to take stereo and rgbd as input, but i dont see any way supplying rgbd input for **demo.py** . Is there any way of doing so ?

felipesce, 2022-11-11:
https://github.com/princeton-vl/DROID-SLAM/issues/45#issuecomment-1135551368

SuruchiPokhrel, 2022-11-18:
I tried the same for my depth input from Kinect camera, but did not really see any pointclouds during reconstruction, and also the camera trajectory seems really off. What is the source of your depth input?

zovelsanj, 2022-11-18:
try this:

`depth = depth.astype(np.float32) / 5000.0`
`depth = torch.from_numpy(depth).float()[None,None]`
`depth = F.interpolate(depth, (h1, w1), mode='nearest').squeeze()`
`depth = depth[:h1-h1%8, :w1-w1%8]`

closed 2022-11-21 by SuruchiPokhrel