princeton-vl/DROID-SLAM #85: Scale problem of droid-slam

gyes00205, 2022-12-25:
Hi authors,
Thanks for your amazing work. I have a question about scale of camera pose.
I use ZED camera to collect data and compare camra pose  of ZED and droid-slam.
Blue dot means camera pose of ZED and red dot means camera pose of droid-slam.

[image]

I find some scale problems of droid-slam. Could you tell me how to solve the scale problem of camera pose?
Thanks.

gyes00205, 2023-12-07:
Hi @kg0221, you can refer to this [reply ](https://github.com/princeton-vl/DROID-SLAM/issues/90#issuecomment-1518675513).

kg0221, 2023-12-20:
Thanks @gyes00205. But I'm not using a stereo camera. So I don't think that solution works for me. However, I did find that the training dataloader scales the world coordinates by 1/5. And the results are not scaled back up afterwords. I'm not sure why this is. But by scaling my inputs by 1/5 and my outputs back up by 5 afterwords, I was able to get my expected results.

lbg030, 2024-05-18:
> Thanks @gyes00205. But I'm not using a stereo camera. So I don't think that solution works for me. However, I did find that the training dataloader scales the world coordinates by 1/5. And the [...]

Hi, I encounter with same problem.
I want to set GT pose, only evaluating 3D reconstruction.
Can you explain how to do ?
thank you so much

Promethe-us, 2024-09-01:
> Thanks @gyes00205. But I'm not using a stereo camera. So I don't think that solution works for me. However, I did find that the training dataloader scales the world coordinates by 1/5. And the [...]

Hello, I meet the save problem, have you solved it? I set gt w2c for each iter, but depth tend to be worse.