princeton-vl/DROID-SLAM #86: Test TartanAir visual SLAM challenge

letsgofeng, 2022-12-28:
When using the original code and given the droid.pth model, the result of the MH004 sequence of TartanAir visual SLAM monocular challenge is very different from the paper. May I ask that the test parameters of the TartanAir visual SLAM challenge are the same as the ”validate_tartanair.py“ script? (because you did not provide the TartanAir visual SLAM challenge script). 
The following is the specific result:
xf.peng@labhpc75:~/code/DROID-SLAM$ CUDA_VISIBLE_DEVICES=2 ~/anaconda3/envs/droidenvnew/bin/python evaluation_scripts/test_tartanair_challenge.py --datapath=/home/xf.peng/dataset/TartanAirChallenge/ --weights=droid.pth
Performing evaluation on mono/MH004
droid.pth
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 662/662 [02:01<00:00,  5.44it/s]
KFs: 372
################################
Global BA Iteration #1
Global BA Iteration #2
Global BA Iteration #3
Global BA Iteration #4
Global BA Iteration #5
Global BA Iteration #6
Global BA Iteration #7
################################
Global BA Iteration #1
Global BA Iteration #2
Global BA Iteration #3
Global BA Iteration #4
Global BA Iteration #5
Global BA Iteration #6
Global BA Iteration #7
Global BA Iteration #8
Global BA Iteration #9
Global BA Iteration #10
Global BA Iteration #11
Global BA Iteration #12
  ATE scale: 0.965949975203637
{'ate_score': 3.7281677627737584, 'rpe_score': (0.9659354892992258, 4.380713634900417), 'kitti_score': (1.7173907932760222, 0.20961711203211916)}

ckLibra, 2023-02-27:
Hi, I meet the same problem. Have u found the reason?

awarebayes, 2023-03-14:
Guys I do not think there is any loop closure implemented...

ckLibra, 2023-04-11:
> Guys I do not think there is any loop closure implemented...

What does that mean? DROID actually does not include loop closure.

Alex-Beh, 2023-04-17:
@ckLibra Does not include loop closure?

dingangui, 2023-08-09:
> > Guys I do not think there is any loop closure implemented...
> 
> What does that mean? DROID actually does not include loop closure.

hey, the paper says DROID has loop closure~ Have you checked it?

ckLibra, 2023-08-14:
> > Guys I do not think there is any loop closure implemented...
> 
> What does that mean? DROID actually does not include loop closure.

Sorry guys. I apologize for my mistakes. DROID indeed claims the loop clousure in their paper. But in my understanding, there are no explicit loop clousure in current implementation. Maybe the impicit one can be found in backend, just like

ThakurSarveshGit, 2024-08-06:
Usually front-end is responsible for marking loop closure edges but in case of Droid SLAM's front-end, I can confirm it detects no loop closure on my custom dataset. Also, there is no implementation for loop closure detection. Factors(edges) are added in these methods: _add_factors_, _add_neighborhood_factors_ and _add_proximity_factors_ and none of them actually looks for loop closure! So, not sure how exactly this is implemented.

Does someone know how is this implemented as quoted in the paper?

zachteed, 2025-05-01:
Hi, sorry for being very late to this. I can confirm that there was an issue with the MH004 scene. This PR https://github.com/princeton-vl/DROID-SLAM/pull/152 adds evaluation for the TartanAir Test set for the stereo and monocular tracks.

This is the excepted result on MH004 after the update.

[nair_test_mono_MH004.pdf](https://github.com/user-attachments/files/20001338/nair_test_mono_MH004.pdf)

Regarding loop closure what @ThakurSarveshGit said is correct, it is based on reprojection error.

closed 2025-05-05 by zachteed