princeton-vl/DROID-SLAM #90: Model files for stereo and RGB-D

GopiRajuMatta, 2023-01-23:
Hello,

Provided droid.pth file works for mono setup. As mentioned in the paper, DROID-SLAM also can be used with stereo and RGB-D settings. Can you please provide model(.pth) files for stereo and RGB-D settings. 

I want to test the efficacy of this method with other stereo visual odometry methods.

Thank you
Gopi

gyes00205, 2023-02-15:
Hi @GopiRajuMatta ,

The paper means that mono, stereo and RGB-D setting can use the same model weights (droid.pth) to test. 
Therefore, you can use the droid.pth to test the efficiency on stereo setting.

PeteBai, 2023-04-21:
> 

hi @gyes00205  Were you able to evaluate stereo sequence with no scaling issue? I'm having trouble configure the stereo sequence. all the results are not up to scale. I also didn't find a way to define parameters such as stereo baseline.... Thank you

gyes00205, 2023-04-22:
Hello @PeteBai,
I don't evaluated the stereo sequence. To address the scaling issue, I estimate the depth of stereo images using a stereo matching network and then feed the resulting RGB-D images into DROID-SLAM.
Below is a figure comparing monocular with stereo + depth. I used ZED camera to capture stereo images.

[image]