princeton-vl/DROID-SLAM #99: Take real-time footage from gopro camera on demo.py

zapmxl, 2023-03-22:
Hello there 
I'd like edit the demo.py to take the live footage from my camera 
I found the code to make the camera work wirelessly but i don't know how to implement it inside demo.py and make live-SLAM  (since i'm still a novice)
any help would be greatly apreciated

moaazassali, 2023-03-26:
I am facing the same issue. Have you find any solution?

pytholic, 2023-03-29:
@zapmxl @moaazassali Hi. Have you guys found a way to run it with a live camera feed from something like GoPro or a webcam?

zhangjd1029, 2023-06-19:
@pytholic @moaazassali @zachteed
Hello, have you solved it yet? Can you share the code

pytholic, 2023-06-19:
@zhangjd1029 Hi. I don't have access to that codebase anymore, it is private. But it wasn't hard.
You need to change `image_stream` function to use webcam instead of reading from directory. I was also able to send these images to YoloV8 and get realtime segmentation. You can also use some other devices. I was using IP Webcam app on S8 ultra tablet.

zhangjd1029, 2023-06-19:
@pytholic 
Okay, thank you. I will try according to your suggestion.
In addition, I am also using a webcam to do some remote SLAM, but the results are always affected by the camera's pixels and distortion, which may be related to the cheaper camera I use. Do you have this confusion? What brand of webcam are you using?
thanks

pytholic, 2023-06-19:
@zhangjd1029 I would recommend using 1080p cam. I was using Anker webcam.

pytholic, 2023-06-19:
@zhangjd1029 Distortion and low quality will definitely cause some issues.

zhangjd1029, 2023-06-19:
@pytholic 
ok! Thank you