princeton-vl/DROID-SLAM #102: Scale issue on stereo setup

yediluo, 2023-04-21:
HI folks
Thank you so much for making this amazing work opensource. However, I have a question regarding the scaling issue in stereo evaluation. I have attempted to use droidslam to evaluate multiple stereo driving datasets. Unfortunately, most of them have a scaling issue if the estimated trajectory is not scaled with the ground truth. As we are using stereo, shouldn't the scale issue not exist? Additionally, I have not been able to find a location to input the stereo baseline parameters. Could you provide me with some insight on how to solve this stereo scaling issue? or did I miss some important steps here?
  Thank you.

moaazassali, 2023-05-04:
I have a similar problem where after a few seconds into the visualization, the scale becomes much smaller. It starts out well but like after 5 seconds, the new visualization is smaller, and after another 5 seconds or so, the visualization gets smaller too. So overall, it looks like camera is just spiraling down a smaller and smaller environment. Note that this only happens on my own videos. I am pretty sure my calibration values are correct and have performed several calibration tests using OpenCV.

chenjx1005, 2023-06-28:
I check the source code and find this line: https://github.com/princeton-vl/DROID-SLAM/blob/main/droid_slam/geom/projective_ops.py#L105
It seems that the author fix the stereo camera's pose as [-0.1, 0, 0, 0, 0, 0, 1], so that the estimate depth is subject to this scale.

sacchinbhg, 2025-11-07:
Hi, I faced the same issue earlier. The problem is that inside src/droid_kernels.cu, the stereo baseline is hardcoded as 0.1 m. For the TartanAir dataset, which actually has a 0.25 m baseline, the evaluation script evaluation_scripts/test_tartanair.py sets the stereo scale factor to 2.5, since 0.25 / 0.1 = 2.5.

For your own dataset, you can make a similar script and set the scale factor to your baseline divided by 0.1. That should fix the scale mismatch. Hope this helps!