princeton-vl/DROID-SLAM #117: Monocular RGB SLAM with fixed poses

pangbo1997, 2023-12-13:
Thanks for the great work. However, I'm struggling with using the inference procedure of the project on the monocular RGB scene.  To ensure the consistency of the pose and scale, I would like to remain several poses fixed and I modified the [droid_kernels. Cu] file, setting Jacobins to zero if the pose is fixed. But the results were far from except. I have also tried not to update the fixed pose, and it didn't work either. Did I get something wrong? Any suggest will be very helpful, thank you!

closed 2023-12-14 by pangbo1997

lbg030, 2024-05-19:
I have same Problem too.
I set GT pose and modify ba_cuda functions that ignore update poses codes.
But It doesn't work.

Did you solve this problem ?

sus71, 2024-12-16:
did you solve the problem?