princeton-vl/DROID-SLAM #149: Compilation error

ighoshsubho, 2025-03-25:
src/altcorr_kernel.cu(309): error: no suitable conversion function from "const at::DeprecatedTypeProperties" to "c10::ScalarType" exists
    [&] { const auto& the_type = fmap1.type(); constexpr const char* at_dispatch_name = "altcorr_forward_kernel"; at::ScalarType _st = ::detail::scalar_type(the_type); ; switch (_st) { case at::ScalarType::Double: { do { if constexpr (!at::should_include_kernel_dtype( at_dispatch_name, at::ScalarType::Double)) { if (!(false)) { ::c10::detail::torchCheckFail( __func__, "src/altcorr_kernel.cu", static_cast<uint32_t>(309), (::c10::detail::torchCheckMsgImpl( "Expected " "false" " to be true, but got false.  " "(Could this error message be improved?  If so, " "please report an enhancement request to PyTorch.)", "dtype '", toString(at::ScalarType::Double), "' not selected for kernel tag ", at_dispatch_name))); }; } } while (0); using scalar_t [[maybe_unused]] = c10::impl::ScalarTypeToCPPTypeT<at::ScalarType::Double>; return ([&] { altcorr_forward_kernel<scalar_t><<<blocks, threads>>>( fmap1.packed_accessor32<scalar_t,4,torch::RestrictPtrTraits>(), fmap2.packed_accessor32<scalar_t,4,torch::RestrictPtrTraits>(), coords.packed_accessor32<float,5,torch::RestrictPtrTraits>(), corr.packed_accessor32<scalar_t,5,torch::RestrictPtrTraits>(), radius); })(); } case at::ScalarType::Float: { do { if constexpr (!at::should_include_kernel_dtype( at_dispatch_name, at::ScalarType::Float)) { if (!(false)) { ::c10::detail::torchCheckFail( __func__, "src/altcorr_kernel.cu", static_cast<uint32_t>(309), (::c10::detail::torchCheckMsgImpl( "Expected " "false" " to be true, but got false.  " "(Could this error message be improved?  If so, " "please report an enhancement request to PyTorch.)", "dtype '", toString(at::ScalarType::Float), "' not selected for kernel tag ", at_dispatch_name))); }; } } while (0); using scalar_t [[maybe_unused]] = c10::impl::ScalarTypeToCPPTypeT<at::ScalarType::Float>; return ([&] { altcorr_forward_kernel<scalar_t><<<blocks, threads>>>( fmap1.packed_accessor32<scalar_t,4,torch::RestrictPtrTraits>(), fmap2.packed_accessor32<scalar_t,4,torch::RestrictPtrTraits>(), coords.packed_accessor32<float,5,torch::RestrictPtrTraits>(), corr.packed_accessor32<scalar_t,5,torch::RestrictPtrTraits>(), radius); })(); } case at::ScalarType::Half: { do { if constexpr (!at::should_include_kernel_dtype( at_dispatch_name, at::ScalarType::Half)) { if (!(false)) { ::c10::detail::torchCheckFail( __func__, "src/altcorr_kernel.cu", static_cast<uint32_t>(309), (::c10::detail::torchCheckMsgImpl( "Expected " "false" " to be true, but got false.  " "(Could this error message be improved?  If so, " "please report an enhancement request to PyTorch.)", "dtype '", toString(at::ScalarType::Half), "' not selected for kernel tag ", at_dispatch_name))); }; } } while (0); using scalar_t [[maybe_unused]] = c10::impl::ScalarTypeToCPPTypeT<at::ScalarType::Half>; return ([&] { altcorr_forward_kernel<scalar_t><<<blocks, threads>>>( fmap1.packed_accessor32<scalar_t,4,torch::RestrictPtrTraits>(), fmap2.packed_accessor32<scalar_t,4,torch::RestrictPtrTraits>(), coords.packed_accessor32<float,5,torch::RestrictPtrTraits>(), corr.packed_accessor32<scalar_t,5,torch::RestrictPtrTraits>(), radius); })(); } default: if (!(false)) { ::c10::detail::torchCheckFail( __func__, "src/altcorr_kernel.cu", static_cast<uint32_t>(309), (::c10::detail::torchCheckMsgImpl( "Expected " "false" " to be true, but got false.  " "(Could this error message be improved?  If so, " "please report an enhancement request to PyTorch.)", '"', at_dispatch_name, "\" not implemented for '", toString(_st), "'"))); }; } }()
                                                                                                                                                             ^

1 error detected in the compilation of "src/altcorr_kernel.cu".
error: command '/usr/local/cuda/bin/nvcc' failed with exit code 2

The error is occurring in the kernel compilation process. I think issue is a type conversion problem, the code is trying to convert const at::DeprecatedTypeProperties to c10::ScalarType, but there's no suitable conversion function available.

kuanhenglin, 2025-04-10:
I encountered the same problem when I'm trying to get another repo which has compiling DROID-SLAM as part of getting it to work, and the solution is to downgrade your PyTorch to be below 2.6.0. From [here](https://github.com/NVIDIAGameWorks/kaolin/issues/865) it seems like PyTorch 2.6.0 deprecated `.type()` and all instances of it should be changed to `.scalar_type()`, and DROID-SLAM uses .type() which results in the deprecated type as shown in the error.

zachteed, 2025-05-03:
I just updated the lietorch and DROID-SLAM code to use .scalar_type() instead of .type() and tested with torch 2.7. This should resolve this issue.

closed 2025-05-07 by zachteed