princeton-vl/DROID-SLAM #156: Rotation accuracy?

jlazarow, 2025-05-08:
Perhaps a basic question here, but the evaluation scripts seem to only show translation accuracy (ATE). Looking into measuring rotation accuracy (i.e., PoseRelation.rotation_angle_deg in evo) generally gives me quite poor results e..g, `131.92` degrees of error on ETH3D-SLAM/desk_3. Am I missing something, is this some sort of coordinate convention that the SE3 alignment cannot capture?

Thanks!

zachteed, 2025-05-09:
Just checked this, you're correct there was an issue for evaluating rotations (This doesn't impact ATE metrics in any way)

`evo` expects quaternions (w, x, y, z) and droid slam outputs (x, y, z, w), so this line needs to be updated for performing evaluation on rotations

```
    traj_est = PoseTrajectory3D(
        positions_xyz=traj_est[:,:3],
        orientations_quat_wxyz=traj_est[:,[6, 3, 4, 5]], # <- make this change here
        timestamps=np.array(tstamps))
```

After this change you should see these metrics for rotations (in degrees)
```
       max	0.014514
      mean	0.005337
    median	0.005137
       min	0.000954
      rmse	0.005698
       sse	0.062763
       std	0.001996

```

jlazarow, 2025-05-09:
Thank you. This indeed fixes it!

closed 2025-05-09 by jlazarow